rerun-io/rerun #5092: View-viewer crashes in Debug build on launch

jleibs, 2024-02-07:
[image]

```
re_viewer.js:648 panicked at /home/jleibs/.cargo/registry/src/index.crates.io-6f17d22bba15001f/once_cell-1.19.0/src/lib.rs:1164:13:
assertion failed: self.0.is_initialized()

Stack:

Error
    at imports.wbg.__wbg_new_17c866ad40da987b (http://localhost:9090/re_viewer.js:654:21)
    at eframe::web::panic_handler::PanicSummary::new::hed77e50ce8fec231 (wasm://wasm/062347c6:wasm-function[15021]:0xf8dd28)
    at eframe::web::panic_handler::PanicHandler::install::{{closure}}::ha6bdda355bf75c3d (wasm://wasm/062347c6:wasm-function[8269]:0xcb4c42)
    at std::panicking::rust_panic_with_hook::hc20eadded6bfe687 (wasm://wasm/062347c6:wasm-function[18940]:0x1092c65)
    at std::panicking::begin_panic_handler::{{closure}}::h82415fe35b0e2001 (wasm://wasm/062347c6:wasm-function[21237]:0x11066ed)
    at std::sys_common::backtrace::__rust_end_short_backtrace::h71f504d46a203d88 (wasm://wasm/062347c6:wasm-function[34279]:0x11d1e47)
    at rust_begin_unwind (wasm://wasm/062347c6:wasm-function[24664]:0x1178ece)
    at core::panicking::panic_fmt::h7a368385936888dc (wasm://wasm/062347c6:wasm-function[26532]:0x119c73e)
    at core::panicking::panic::h7bbea3773b752235 (wasm://wasm/062347c6:wasm-function[25879]:0x1191c67)
    at re_space_view_spatial::spatial_topology::SpatialTopology::access::h6270f5b70e7084a1 (wasm://wasm/062347c6:wasm-function[17747]:0x104d5dc)
imports.wbg.__wbg_error_fef71bc9b670aec4 @ re_viewer.js:648
062347c6:0x11d3222 Uncaught RuntimeError: unreachable
    at __rust_start_panic (062347c6:0x11d3222)
    at rust_panic (062347c6:0x11cc35e)
    at std::panicking::rust_panic_with_hook::hc20eadded6bfe687 (062347c6:0x1092c95)
    at std::panicking::begin_panic_handler::{{closure}}::h82415fe35b0e2001 (062347c6:0x11066ed)
    at std::sys_common::backtrace::__rust_end_short_backtrace::h71f504d46a203d88 (062347c6:0x11d1e47)
    at rust_begin_unwind (062347c6:0x1178ece)
    at core::panicking::panic_fmt::h7a368385936888dc (062347c6:0x119c73e)
    at core::panicking::panic::h7bbea3773b752235 (062347c6:0x1191c67)
    at re_space_view_spatial::spatial_topology::SpatialTopology::access::h6270f5b70e7084a1 (062347c6:0x104d5dc)
    at <re_space_view_spatial::space_view_2d::SpatialSpaceView2D as re_viewer_context::space_view::space_view_class::SpaceViewClass>::spawn_heuristics::h3325890ba9f3f3ac (062347c6:0xa7a8cb)
```

teh-cmc, 2024-02-07:
I cannot repro in either debug or release; does it need to be launched in a specific way..?

jleibs, 2024-02-07:
`just rerun-web` was sufficient for me when I reported it but now it's not repro'ing for me either. Heisenbug?

Wumpf, 2024-02-13:
closing until we get hold of it again

closed 2024-02-13 by Wumpf