rerun-io/rerun #5095: Compile times

v-morlock, 2024-02-07:
**Describe the annoyance**
Compile times for rerun are rather long right now. Even when disabling most of its features, its compile-time really adds up and i'm wondering whether things like egui really need to be compiled when we're only using the sdk to log data..

We have a medium-size rust application and in the compile-timings, the crates with the longest compile-times all seem to be rerun-related.

This is how we add rerun in our cargo.toml
rerun = { version = "0.12.1", default-features = false, features = ["sdk"] }

**To Reproduce**
Install rerun, compile with `cargo build --timings`

**Screenshots**
[image]

emilk, 2024-02-07:
I agree completely. However, I think this existing issue covers it
* https://github.com/rerun-io/rerun/issues/4788

I'll add some more notes on that one shortly.

closed 2024-02-07 by emilk

Wumpf, 2024-02-07:
@emilk something strange is going on there though. Why are we building with egui when not including the viewer itself

Wumpf, 2024-02-07:
ah you already noted that on the other issue now 👍

emilk, 2024-02-07:
…and solved it: https://github.com/rerun-io/rerun/pull/5099