rerun-io/rerun #5100: Unknown container popup + warning when dragging tile

roym899, 2024-02-07:
**Describe the bug**
See this video:

https://github.com/rerun-io/rerun/assets/9785832/7373f7d9-6955-490c-9c68-9072a98e2f73

**To Reproduce**
See video. Also reproducible from examples. Also getting a warning 
> Container for tile ID ... not found

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04

**Rerun version**
> rerun_py 0.13.0-alpha.3 [rustc 1.74.0 (79e9716c9 2023-11-13), LLVM 17.0.4] x86_64-unknown-linux-gnu main 79d8021, built 2024-02-07T15:07:02Z

abey79, 2024-02-07:
This appears to be the offending piece of code:

https://github.com/rerun-io/egui_tiles/blob/c2dae1eaa27ba1a4e92fa538b98b112b38db97e7/src/container/tabs.rs#L271-L285

When commented out, it becomes impossible to drag from the "blank" space (the tile title UI must be dragged instead). This seems the right thing to do, as this blank space should rather clear the selection when clicked IMO.

Still need to figure out the details of that code to have a fix.

emilk, 2024-02-07:
Right, you are dragging a _container_, and not a space-view, and something is going wrong when fetching the title for that. We should probably fix this before the release if we can.

emilk, 2024-02-07:
I'd say this is the problematic code:

https://github.com/rerun-io/rerun/blob/1ddc42a131d5dd6b4907b1836e24f0158c1af1a5/crates/re_viewport/src/viewport.rs#L927

Why is the dragged tile-id not properly matched back to a blueprint container?

On failure, we can also show a much nicer placeholder: "Tab Container" in nice text, for instance.

abey79, 2024-02-07:
Maybe because those are single-child Tab containers, created on the fly to "force" the tile tab UI to show up? These don't have matching blueprint containers believe.

abey79, 2024-02-08:
On second thought we should be able to detect single child situation and look up the enclosed space view.

commit 0048f236 (2024-02-08): Fix display issue when dragging space views from the tile bar blank space (#5119)

### What

- Fixes #5100

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested the web demo (if applicable):
* Using newly built examples:
[app.rerun.io](https://app.rerun.io/pr/5119/index.html)
* Using examples from latest `main` build:
[app.rerun.io](https://app.rerun.io/pr/51 [... cut]
closed 2024-02-08 by abey79