rerun-io/rerun #5157: Advertised frame time at rest much higher for WebGPU vs. WebGL in Chrome with nuscenes

abey79, 2024-02-09:
The advertised frame time in WebGPU is consistently higher than in WebGL mode _at rest_:

[image]
[image]

When scrolling/playing around both seems similarly smooth. When playing the recording in a loop, WebGPU frame time (very subjectively) appear smaller on average.

Chrome Mac
Version 121.0.6167.160 (Official Build) (arm64)

emilk, 2024-02-09:
Build the web viewer in release mode with debug symbols, then look at the in-browser profiler for clues

abey79, 2024-02-09:
How do you get symbols? I tried this but apparently isn't sufficient.

```
[profile.release]
debug = true # good for profilers
```

[image]

abey79, 2024-02-09:
These call explain a fair chunk of the apparent timing behaviour.

[image]

In contrast, with WebGL, very little time is taken by the time `re_renderer` is reached:
[image]

That's about as deep an analysis I'm able to produce here. @Wumpf will have much more insight about all this.