rerun-io/rerun #5206: Stopping to track camera no longer resets up vector

Wumpf, 2024-02-15:
Likely a regression from

* #5193

When stopping to track a camera, we don't reset the up vector:

https://github.com/rerun-io/rerun/assets/1220815/c07695eb-c539-4930-a7ba-9e2d152db8fd

emilk, 2024-02-16:
I consider this a feature, not a bug. You move into the camera space, and see the world as the camera sees it. If you don't want to do that, double-click to reset. We could of-course also add a button to reset only the up-axis.

Wumpf, 2024-02-16:
Hmm I found it confusing, but yeah my natural instinct was double clicking somewhere.
I figure this might get better once we have blueprint Eye controls where you can see these settings then

closed 2024-02-16 by Wumpf