rerun-io/rerun #5268: Can't receive larger WebSocket messages than 16MiB on native

emilk, 2024-02-23:
If a users tries to send a big chunk of that (e.g. a 10 mega-pixel RGBA image) then they might hit the default 16 MiB data frame limit in tungstenite:

https://docs.rs/tungstenite/latest/tungstenite/protocol/struct.WebSocketConfig.html#structfield.max_frame_size

We should increase this maximum by a lot, which will require changes to [ewebsock](https://github.com/rerun-io/ewebsock)

mpizenberg, 2024-02-23:
Is this error likely to happen if I transfer a large mesh too? If increasing the max frame size is problematic, would it make sense to instead chunk the data sent into blocks smaller than the max frame size?

mpizenberg, 2024-02-26:
I can confirm that loading a medium size mesh crashes the viewer too. I have meshes of sizes up to 1GB approximately that I'd need to visualize eventually.

closed 2024-02-26 by Wumpf