rerun-io/rerun #5298: New options for setting blueprint: "force", "load if unset", "set default"

jleibs, 2024-02-26:
Builds on top of:
 - https://github.com/rerun-io/rerun/issues/5297

Now that a blueprint can be set at any time, we want some minimal control for the SDK for how this behaves.

Force and load-if-unset are mostly trivial and could likely be added to the "ActivateBlueprint" message proposed in (https://github.com/rerun-io/rerun/issues/5297)

"Set default" includes some extensions to the StoreHub.
 - The basic idea is that the store-hub will store **two** blueprints per app-id.  One is the currently active blueprint, where all user edits are applied. The other is our first implementation of a "snapshot" as the MVP of a single "default" snapshot per blueprint.

A blueprint sent from the SDK will have the option to only replace the "default" snapshot, and a new variant of the "Reset blueprint" action will be to reset to the blueprint to the (SDK-provided) default as opposed to the (Heuristic provided) default.

emilk, 2024-03-25:
Very related to
* #5629

jleibs, 2024-03-25:
Proposal for 0.15: Always use "load if unset + set default"

It's 1 click for a user to reset.

jleibs, 2024-03-27:
Considering this done as of: https://github.com/rerun-io/rerun/pull/5678

closed 2024-03-27 by jleibs