rerun-io/rerun #5305: Arrow deserialization should assume, not check

teh-cmc, 2024-02-27:
In the past we've made the mistake of checking that the incoming datatype matched exactly the expected datatype.
This leads to major slowdowns for little to no gains: we already run SDK version checks anyhow (and if the user is manually serializing stuff, the responsibility to get things correctly is on them).

An example of this is carefully looking through validity bitmap values even though we expect non-nullable data. We should just assume that the data is non-nullable and not look at the bitmap at all.

We've already reverted course on some specific paths (e.g. vec3), leading to gigantic performance wins, but this should really be the default in all cases: interpret the data as if it's correct -- akin to a cast -- as opposed to checking for correctness.
Of course, if the data is so far from what we expect it to be that it cannot even fit into what we're trying to cast it as, we would still fail with an error.

emilk, 2024-02-27:
To clarify: we should focus on performance during deserialization, but still handle any data without crashing.

However, I think we should still do some datatype validation during ingestion.
That would help us diagnose problems from version mismatches that would otherwise become very hard to diagnose.
For instance: changing the order of enum variants should lead to at least a warning, not just a silent bug.
This validation would only run once per component per micro-batch (log message), so it should not affect performance adversely.

teh-cmc, 2024-02-28:
Another related mistake: our serialization APIs are too flexible, allowing to mix references and owned data (`Iterator<Cow<T>>`).

While I haven't seen any performance issues from this for now, it still makes for complicated code for little to no gains.

The `Cow` should be "on the outside", i.e. `Either<Iterator<&T>, Iterator<T>>`

jleibs, 2024-03-06:
Tangentially related, but somewhat relevant to where and how we check things, we have so many different ways for a component to be missing from query results:
- Not logged ever
- Not logged on this timeline
- Not logged on this timeline at this timepoint
- Logged as [],
- Logged as [Null]
- Logged as data that couldn't be deserialized
- Logged as data that can be deserialized, but with a schema that doesn't match the expected schema for the component