rerun-io/rerun #5332: Reduce CI costs

emilk, 2024-02-28:
GitHub actions is costing us more and more each month. Time to rein it it a bit.

We should consider:
* Move jobs from `PR` to `main`
* Move jobs from `main` to `nightly`
* Only run jobs if needed (e.g. ignore Rust jobs when on docs changes)
* Look for redundancies (are we building the web viewer multiple times per PR? 🤷 )
* Consider not running the full CI on each commit
* Move away from GitHub actions

teh-cmc, 2024-02-28:
Any feature permutation will wipe the cargo cache, which means cargo will rebuild the entire workspace from scratch any time that happens.
I wouldn't be surprised if we do so at least ten times for each CI run right now (every different clippy/build/test/doc invocations, every different SDKs, etc), probably even more.

At the very least, the PR CI should only build the most important feature permutations, and keep everything else for the `main` branch.

teh-cmc, 2024-02-28:
We should also try cheaper runners, which would also cover our arm64 needs.

chengguizi, 2024-03-01:
Also, seems that the C++ part always build examples and docs
https://github.com/rerun-io/rerun/blob/1643db8118fabe901435e9f8d952aa083837503d/CMakeLists.txt#L158-L160

would be great if that can be a cmake option. So building CI/CD as someone using rerun saves time

Wumpf, 2024-03-01:
@chengguizi this means that a single CMake's configure always knows of all the targets, but we're not necessarily building all the targets on every CI step