rerun-io/rerun #5352: Visual glitch in blueprint tree UI for hidden item

abey79, 2024-02-29:
[image]

Introduced in:
- https://github.com/rerun-io/rerun/pull/5219

emilk, 2024-02-29:
I fail to reproduce. If I hover an item it looks fine:

[image]

abey79, 2024-02-29:
@emilk my bad, I forgot to specify that this happens only when the item is hidden

abey79, 2024-03-07:
Likely for the same reason, the rerun menu has the same glitches:
[image]

Wumpf, 2024-03-11:
the later is 
* #5368

emilk, 2024-03-12:
Duplicate of [#5368](https://github.com/rerun-io/rerun/issues/5368)

commit 66bd58bc (2024-03-13): Update to egui master (#5486)

### What
* Closes #5352
* Closes #5368

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested the web demo (if applicable):
* Using newly built examples:
[app.rerun.io](https://app.rerun.io/pr/5486/index.html)
* Using examples from latest `main` build:
[app.rerun.io](https://app.rerun.io/pr/5486/index.html?manifest_url=https://ap [... cut]
closed 2024-03-13 by emilk