rerun-io/rerun #5391: Visualization of multiple sources of point clouds does not work

changh95, 2024-03-05:
**Describe the bug**

I'm comparing two different lidar scans (pre- vs post-processed lidar point clouds) that are on the same timestamps.

For each lidar scan, I've assigned a unique entity. (/lidar/scan, /lidar/deskew).

I'm expecting to see the two lidar scans drawn in the visualizer at the same time, but the visualizer fails.

I have seen many different types of errors. 

So far I've observed 

- None of the entities being drawn.
- Only one of the entities being drawn.
- Both are drawn, but all disappear when I change frame
- ... and other complex behaviours

Below is the part where I log the two lidar scans

```
import rerun as rr
import numpy as np
import os
import time
from pyntcloud import PyntCloud

rr.init("lidar_scan_viewer", spawn=True)

rr.spawn()
rr.set_time_seconds("stable_time", 0)

# ...

i = 0
for file1, file2 in zip(files1, files2):
    cloud1 = PyntCloud.from_file(os.path.join(path1, file1))
    xyz1 = cloud1.xyz
    intensity1 = cloud1.points["intensity"].values
    timestamps1 = cloud1.points["timestamp"].values

    # Log the entire point cloud
    rr.set_time_seconds("stable_time", i)
    rr.log(
        "lidar/scan",
        rr.Points3D(
            xyz1,
            colors=[1.0, 1.0, 1.0]
        ),
    )
    
    time.sleep(0.01)

    cloud2 = PyntCloud.from_file(os.path.join(path2, file2))
    xyz2 = cloud2.xyz
    intensity2 = cloud2.points["intensity"].values
    timestamps2 = cloud2.points["timestamp"].values

    rr.set_time_seconds("stable_time", i)
    rr.log(
        "lidar/deskew",
        rr.Points3D(
            xyz2,
            colors=[0.5, 0.5, 0.5]
        ),
    )
    time.sleep(0.01)
    
    i+=1
```

**To Reproduce**
Steps to reproduce the behavior:

Refer to the code above.

1. Run the Python script
2. Rerun app opens. Check for the visualization.
3. See error

**Expected behavior**

Both lidar scans being drawn at the same timestamp.

**Screenshots**

- When only logging single-source point cloud (Just to prove that the point cloud is not corrupted)
[image]

- When logging point clouds from 2 sources (/lidar/scans (White), /lidar/deskew (Gray))
   - When the data is loaded, no point cloud is visible
   - Double clicking on /lidar/scans (white) makes /lidar/deskew (gray) visible
   - When 'play' button is pressed, suddenly /lidar/scans (white) is visible and visualized in sequence
   - When I toggle visibility of /lidar/scans (white) to off while /lidar/deskew (gray) is on, no point cloud is visible
   - Double clicking on /lidar/deskew (gray) suddenly makes the point cloud visible, though when zoomed in, the UI breaks, and when played, the point cloud disappears again.
[image]

**Backtrace**

**Desktop (please complete the following information):**

Ubuntu 22.04 64bit

**Rerun version**

rerun_py 0.14.1 [rustc 1.74.0 (79e9716c9 2023-11-13), LLVM 17.0.4] x86_64-unknown-linux-gnu release-0.14.1 74f1c23, built 2024-02-29T11:00:55Z

**Additional context**

changh95, 2024-03-05:
Nevermind!
One of the point cloud was heavily corrupted.

rerun log function works perfectly well. 

Sorry for the confusion!

Closing the issue.

closed 2024-03-05 by changh95

Wumpf, 2024-03-05:
@changh95  Happy you figured it out, but I'd love to learn what caused Rerun to break like this! In what ways was your point cloud broken? From the looks of it I'd guess that the scale was off the charts, causing floating point inaccuracies, but I'm not sure.