rerun-io/rerun #5398: Viewer navigation

jprochazk, 2024-03-05:
We currently don't handle back/forward navigations in any way.

In practice, this means that:
- A user sees a link to our [web app](https://app.rerun.io), and clicks it
- Scrolls down to the example section, and clicks on one of them
- Looks at it for a bit
- Navigates backward (either mouse 5 or a swipe right)
This puts them back to where they were in the browser before navigation to [app.rerun.io](https://app.rerun.io). That's quite a jarring experience.

The back button in the scenario above should've taken the user back to the welcome screen, while also preserving the scroll position. They may then either:
1. Open a different example.
2. Navigate forward (mouse 4/swipe left) to go back to the example they opened previously.

We should implement a navigation history within the viewer, mirroring the web [history API](https://developer.mozilla.org/en-US/docs/Web/API/History_API).
What exactly requires an entry in the navigation history is somewhat unclear. We can start with only keeping track of navigations between the welcome screen and a recording, and between individual recordings when multiple are open at the same time.

We may eventually want to expand what we keep track of in the navigation history to more than just the welcome screen and recordings. For example, in Slack, clicking on "replies" under a post thread will "split" the main content area into the channel and reply thread (as long as there is enough space for both), where navigating back will close the thread, and subsequently navigating forward will re-open the thread.

nikolausWest, 2024-03-11:
For the simple case here, would we drop the recording if you click and example on the welcome page, wait for recording to load, press back?

jprochazk, 2024-03-25:
As a first step, we should handle only navigations between the example page and recordings:
- Opening a recording from the example page set a "navigated from example page" flag.
- Pressing the back button when the "navigated from example page" flag is set will _close all open recordings_, effectively returning the user back to the example page.
- Pressing the back button when the "navigated from example page" flag is _not_ set will fall back to the browser default behavior, and on native do nothing.

This may introduce weird edge cases, but we're okay with that, because we just want the happy path of onboarding to work.

nikolausWest, 2024-03-25:
This is "high" urgency if it's a quick fix. Otherwise it's "medium"

emilk, 2024-03-28:
We can also use the strategy of egui.rs and add `#fragments` to the url which is a real part of the history

emilk, 2024-04-02:
It seems like the best course of action would be to use the [history API](https://developer.mozilla.org/en-US/docs/Web/API/History_API), as Jan suggested.

When _leaving_ the examples/welcome page (clicking an example) we can call both:
```js
history.pushState({}, "", "?examples");
history.pushState({}, "", "?url=…");
```

Then when the user hits the back-button, they will end up at `?examples` (which we would need to parse and handle). When they press the forward button again, they end up at `url=…` which will load the example again.

We would need to install an even handler to listen for these changes with

```js
window.addEventListener('popstate', function(event) {
   …
})
```

closed 2024-04-03 by emilk