rerun-io/rerun #5414: pinhole.hpp compiler error with std::max

rgolovanov, 2024-03-06:
**Describe the bug**
Compiler error pointing to std::max function in [pinhole.hpp](https://github.com/rerun-io/rerun/blob/0.14.1/rerun_cpp/src/rerun/archetypes/pinhole.hpp#L156)
```
const float focal_length_y = 0.5f / std::tan(std::max(fov_y * 0.5f, EPSILON));
```
Compiler errors:
```
[build] XXX\rerun\archetypes/pinhole.hpp(156,63): error C2589: '(': illegal token on right side of '::' (compiling source file YYY.cpp) [ZZZ.vcxproj]
[build] XXX\rerun\archetypes/pinhole.hpp(156): error C2144: syntax error: 'unknown-type' should be preceded by ')' (compiling source file YYY.cpp) [ZZZ.vcxproj]
[build] XXX\rerun\archetypes/pinhole.hpp(156,1): error C2661: 'tan': no overloaded function takes 0 arguments (compiling source file YYY.cpp) [ZZZ.vcxproj]
[build] XXX\rerun\archetypes/pinhole.hpp(156,25): error C2737: 'focal_length_y': const object must be initialized (compiling source file YYY.cpp) [ZZZ.vcxproj]
[build] XXX\rerun\archetypes/pinhole.hpp(156): error C2144: syntax error: 'unknown-type' should be preceded by ';' (compiling source file YYY.cpp) [ZZZ.vcxproj]
[build] XXX\rerun\archetypes/pinhole.hpp(156,63): error C2143: syntax error: missing ')' before '*' (compiling source file YYY.cpp) [ZZZ.vcxproj]
[build] XXX\rerun\archetypes/pinhole.hpp(156,63): error C2143: syntax error: missing ';' before '*' (compiling source file YYY.cpp) [ZZZ.vcxproj]
[build] XXX\rerun\archetypes/pinhole.hpp(156,1): error C2082: redefinition of formal parameter 'fov_y' (compiling source file YYY.cpp) [ZZZ.vcxproj]
[build] XXX\rerun\archetypes/pinhole.hpp(156,1): error C2059: syntax error: ')' (compiling source file YYY.cpp) [ZZZ.vcxproj]
[build] XXX\rerun\archetypes/pinhole.hpp(156,63): error C2100: illegal indirection (compiling source file YYY.cpp) [ZZZ.vcxproj]
[build] Build finished with exit code 1
```

**To Reproduce**
Steps to reproduce the behavior:
1. Compile C++ project linking Rerun

**Expected behavior**
No compiler errors

**Desktop (please complete the following information):**
 - OS: Windows, VS 2019

**Rerun version**
0.14.1

**Additional context**
Workaround is manually patch this line by adding braces:
```
const float focal_length_y = 0.5f / std::tan((std::max)(fov_y * 0.5f, EPSILON));
```

Wumpf, 2024-03-06:
The problem is that you include `windows.h` before including rerun. Windows has macros for `min` and `max`, imho the better solution is to add  `#define NOMINMAX` before including `windows.h`.
Putting `(std::max)` in parenthesis is an interesting solution I haven't thought of before, we can add that in!

rgolovanov, 2024-03-06:
@Wumpf you are right. It is coming from `windows.h` and `NOMINMAX` resolves the issue.

Wumpf, 2024-03-08:
went with @nasteyi suggestion and added a test to validate rerun.hpp on our ci for this. Fixed in next release then!

PR #5432 (2024-03-08): Make pinhole.hpp robust against min/max preprocessor macros (typically from windows.h)
### What

And add a test to ensure we don't run into this again!

* Fixes #5414

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and the [Code of Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested the web demo (if applicable):
  * Using newly built examples: [app.rerun.io](https://app.rerun.io/pr/5432/index.html)
  * Using examples fro [... cut]
commit 5cd2b6ac (2024-03-08): Make pinhole.hpp robust against min/max preprocessor macros (typically from windows.h) (#5432)

### What

And add a test to ensure we don't run into this again!

* Fixes #5414

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested the web demo (if applicable):
* Using newly built examples:
[app.rerun.io](https://app.rerun.io/pr/5432/index.html)
* Using example [... cut]
closed 2024-03-08 by Wumpf