rerun-io/rerun #5421: Split `ViewportBlueprint` in its own crate

abey79, 2024-03-07:
Attempting to add support context menu in the streams tree (`re_time_panel`) prompted a realisation that the data structure (and related operation) collectively implemented by `{Viewport|Container|SpaceView}Blueprint` are of much wider significance than _just_ displaying the viewport and related tree.

For example, right-clicking on a streams' entity to add it to a new space view is a meaningful operation that implies a dependency of the streams view on the viewport blueprint.

After a quick glance, it appears that the following files/structs could be relatively painlessly be split of in a new `re_viewport_blueprint` (name TBD):
- `ViewportBlueprint` (all of `viewport_blueprint.rs`)
- `ContainerBlueprint` (all of `container.rs`)
- all codegend objects (`blueprint/*`
- `TreeAction`

Assuming the context menu code is further split off in yet another new crate, this would result in the following hierarchy (box may depend on lower box but **not** same-level box):

```
┌───────────────────────────────────────────────┐
│                   re_viewer                   │
└───────────────────────────────────────────────┘
┌──────────────────────┐ ┌──────────────────────┐
│     re_viewport      │ │    re_time_panel     │
└──────────────────────┘ └──────────────────────┘
             ┌──────────────────────┐            
             │   re_context_menu    │            
             └──────────────────────┘            
┌──────────────────────┐ ┌──────────────────────┐
│re_viewport_blueprint │ │      re_data_ui      │
└──────────────────────┘ └──────────────────────┘
             ┌──────────────────────┐            
             │    re_space_view     │            
             └──────────────────────┘            
┌───────────────────────────────────────────────┐
│              re_viewport_context              │
└───────────────────────────────────────────────┘
```

Stretch goal: I haven't deeply looked into that, but, given this refactor, it _might_ be possible to further split off the blueprint tree UI in a separate crate. At the very least, `viewport_blueprint_ui.rs` should be renamed `viewport_tree_ui.rs`.

abey79, 2024-03-12:
This would help (or even unblock):
- #4887

teh-cmc, 2024-04-11:
The codegen'd types have been split into a dedicated crate (`re_types_blueprint`), `re_viewport_blueprint` still has to be done.

closed 2024-05-23 by abey79