rerun-io/rerun #5435: Turn all in-repo python examples into mini-tutorials

nikolausWest, 2024-03-08:
After https://github.com/rerun-io/rerun/issues/5434, add the "Logging and Visualizing with Rerun" section to all example READMEs, in effect turning them into mini-tutorials on how to use Rerun.

BirgerMoell, 2024-03-13:
I'll get started with the controlnet demo.

BirgerMoell, 2024-03-13:
@nikolausWest first draft for a updated controlnet demo.
https://github.com/rerun-io/rerun/pull/5490

andreasnaoum, 2024-03-14:
@BirgerMoell Hey man, I suggest using my Notion page to document what we have done and what is pending.

Here's a public link where you can edit online:
https://rerunio.notion.site/READMEs-15f2e82ab1e44b7cb62e91209ae6c8b2?pvs=4

I started updating them firstly with Intro, Rerun Types and Run sections, and later I will add the Background and Logging sections.

andreasnaoum, 2024-03-21:
Updated READMEs:

Detect and Track Objects
Dicom MRI
Face Tracking
Gesture Detection
Human Pose Tracking
LiDAR

andreasnaoum, 2024-03-22:
#5653  Updated READMEs for the examples:

LLM Embedding-Based Named Entity Recognition
nuScenes
Objectron
Open Photogrammetry Format
Raw Mesh

andreasnaoum, 2024-03-28:
Updated READMEs for the examples:

RGBD
Segment Anything Model
Signed Distance Fields
Structure from Motion
ROS Node (+ setup/dependencies, no explanation just referred to documentation)

andreasnaoum, 2024-04-04:
Updated READMEs for the examples:

Minimal Example
Clock
Multithreading
Multiprocessing
Plots

The last comment and this one is related to this pull request: #5784