rerun-io/rerun #5436: Turn all in-repo C++ examples into mini-tutorials

nikolausWest, 2024-03-08:
After https://github.com/rerun-io/rerun/issues/5434, add the "Logging and Visualizing with Rerun" section to all example READMEs, in effect turning them into mini-tutorials on how to use Rerun.

andreasnaoum, 2024-04-04:
I noticed that C++ examples on the website redirect users to external repositories, and I have a question for the Run the Code section.

These repos also have information on how to run the example.

Should I redirect them to the example repo instead of explaining how to run it there? (see in the final section Run the Code)

[image]

nikolausWest, 2024-04-04:
> Should I redirect them to the example repo instead of explaining how to run it there? (see in the final section Run the Code)

Yes, for all the external ones you can skip the code section and potentially even the "used rerun types" if thats too much work. You can just refer to the repositories for instructions on how to run