rerun-io/rerun #5451: Improve the orbit eye  origin

abey79, 2024-03-11:
One aspect that I consistently found to behave is surprising way is the orbit origin of the 3D view eye, and I have reasons to believe that I'm not alone.

After spending some time in OnShape, I noticed that it uses the bbox center of the geometry **visible on screen** as default orbit point. Also, if an hovered object is (right-)clicked, then it is used as orbit point instead. We should consider implementing something similar.

https://github.com/rerun-io/rerun/assets/49431240/45d5b463-4f7e-4030-9753-4dd31f85b664

https://github.com/rerun-io/rerun/assets/49431240/c6d50db4-72d8-4224-a40a-821640c83ed4

abey79, 2024-03-11:
Using the centre of bounding box of all displayed objects (on screen or otherwise) would already be an improvement IMO.

abey79, 2024-03-11:
Dual cube OnShape playground: https://cad.onshape.com/documents/6183b6e53b67ad1a89aa7abb/w/cbc3cec32eb338f09732aa38/e/702de136749c7ed77764f756?renderMode=0&uiState=65eebcf38b06243496523313

Wumpf, 2024-03-11:
>  I noticed that it uses the bbox center of the geometry visible on screen as default orbit point.

doesn't seem to be the case, once two boxes are on screen it doesn't orbit around the center of those two, only if there's a single one on screen. I'm a bit unclear what it orbits around actually.

Taking our orbit camera model as-is today and change the orbit center also means changing the camera's position, thus completely changing the screen's content. This clearly doesn't happen in onshape, they decouple this somehow further.

What might be related is that the pan operation changes the camera's center point on the image plane, *not* the camera position. Something we don't support at all.

Wumpf, 2024-03-11:
someone in [this thread](https://forum.onshape.com/discussion/4285/can-i-change-the-orbit-mode-to-something-like-sketchup) called this style of orbiting "mcad style", maybe that gives us some pointers to look deeper into this later :)