rerun-io/rerun #5454: New categorization system for examples

jprochazk, 2024-03-11:
The current system is:
- Each example is under a separate category
- Each category is its own tab

The viewer is not consistent with this. The examples there are not categorized in any way, and which ones are present depends on if they can be run in the viewer or not.

We want to move away from the current system, and make the viewer consistent with the website:
- All examples should live on the same page
- We should include all examples, even if they are not runnable
- If an example is runnable, it should be indicated somewhere, and clicking on it will:
 - Start a new inline web viewer on the website
 - Stream the recording directly in the native viewer

Because all examples now live on the same page, we'll need a way to filter them based on their tags.

nikolausWest, 2024-03-11:
The current plan I think is more something like:

## In the gallery on rerun.io/examples:

- Show each example as a card, whether runnable or not
- If runnable, include a quick action button that opens up the viewer and loads it directly
- Clicking on a card opens up a preview / modal of the example details page
- The preview / modal can be expanded to be full page 
- The example detail page has a component that shows 
  - a preview screenshot of the example
  - a link to the source code
  - a description of what it is
  - the Rerun types used
  - (optionally) a mini tutorial with the code snippets with rerun calls used to create it
  - instructions of how to run it
- If the example is runnable (has an available rrd file) there is a button that replaces the preview image with an inline web-viewer
- There is a categorization system that helps users filter and navigate all the examples
  - Current design is based on tags, which are themselves broken into 3 categories
  - We could also go with something simpler, like putting all examples on the same page divided into categories

# In the in-app welcome page

- Shows all runnable examples as cards
- Clicking on a card streams the rrd into the viewer (web or native)
- To begin with there is no filtering or higher level categorization of the examples

nikolausWest, 2024-03-25:
Replaced by:
- https://github.com/rerun-io/rerun/issues/5665

closed 2024-03-25 by nikolausWest