rerun-io/rerun #5455: Heuristics dont create a 3D space view in a simple case

abey79, 2024-03-11:
Consider this code:

```python
import rerun as rr

rr.init("rerun_example_bug")
rr.connect()
rr.set_time_sequence("frame_nr", 0)

rr.log("boxes3d", rr.Boxes3D(centers=[[0, 0, 0], [1, 1.5, 1.15], [3, 2, 1]], half_sizes=[0.5, 1, 0.5] * 3))
rr.log("boxes2d", rr.Boxes2D(centers=[[0, 0], [1.3, 0.5], [3, 2]], half_sizes=[0.5, 1] * 3))
```

Expected: bot a 2D and 3D space view are created
Actual: only a 2D space view is created

[image]

Wumpf, 2024-03-11:
May or may not be related to this user report https://discord.com/channels/1062300748202921994/1075873257124810852/1214908729321988166 on pinhole not getting a 3d scene for
```python
import numpy as np
import rerun as rr

rr.init("rerun_example_pinhole", spawn=True)
rng = np.random.default_rng(12345)

image = rng.uniform(0, 255, size=[3, 3, 3])
rr.log("world/image", rr.Pinhole(focal_length=3, width=3, height=3))
rr.log("world/image", rr.Image(image))```

abey79, 2024-03-11:
Full code:

```
import rerun as rr

rr.init("rerun_example_bug")
rr.connect()
rr.set_time_sequence("frame_nr", 0)

rr.log("boxes3d", rr.Boxes3D(centers=[[0, 0, 0], [1, 1.5, 1.15], [3, 2, 1]], half_sizes=[0.5, 1, 0.5] * 3))
rr.log("boxes2d", rr.Boxes2D(centers=[[0, 0], [1.3, 0.5], [3, 2]], half_sizes=[0.5, 1] * 3))
rr.log("text_logs", rr.TextLog("Hello, world!", level=rr.TextLogLevel.INFO))
rr.log("points2d", rr.Points2D([[0, 0], [1, 1], [3, 2]], labels=["a", "b", "c"]))
```

Various combination of enabled/disabling each of these log lines yield various heuristics results, many of which are currently wrong.

commit b068aa71 (2024-03-18): Fix recommendation from one Space View class affecting recommendation from another (#5554)

### What

This broke recently due to the tracking of space view recommendation
hashes _without_ taking into account their class.
Cleaned up the whole thing a little bit in the progress.

* Fixes #5455
* Also mentioned on this ticket is this issue but it turned out to be
unrelated (thus now a separate ticket) #5553
    * #5455 is new to 0.15alpha, but #5553 happens on 0.14!

---

For this script:
```py
import rerun as rr

rr.init("rerun_example_bug")
rr.connect()
rr.set_time_sequence("frame_nr", 0)

rr.l [... cut]
closed 2024-03-18 by Wumpf