rerun-io/rerun #5467: Make it possible for a user to understand why their blueprint doesn't work

jleibs, 2024-03-11:
At the moment our heuristics generally create space-views that are designed to be functional.

However, Blueprint is likely to introduce many new means of users creating blueprints that don't actually do what the user intended.

For example:
 - Creating query expressions that don't match the entity tree
 - Using an origin that precludes visualizing the added content
 - Overriding components that don't match the visualizer

We need to make it easy and obvious to figure out what's going wrong when things have been explicitly added to a blueprint but don't have any impact.

emilk, 2024-03-25:
Related to:
* https://github.com/rerun-io/rerun/issues/4304

jleibs, 2024-03-25:
Possible small improvements:
 - Better text when nothing is found?
 - When there are **multiple** entity expressions, warn if any single entity expression doesn't contribute content.