rerun-io/rerun #5468: Create blueprints for common examples

jleibs, 2024-03-11:
Creating blueprints for the main examples we include on app.rerun.io has a two-fold benefit.

(1) It allows us to validate the blueprint APIs work for the use-cases covered by these examples
(2) Should improve the clarity of the examples themselves once these blueprints are also served as a part of the examples.

jprochazk, 2024-03-12:
We should also have blueprints for every snippet, at minimum to disable the timeline and blueprint views.

abey79, 2024-03-12:
Related:
- #5477

emilk, 2024-03-18:
On the flipside; if an example works well without a blueprint (e.g. Dicom) then I think it makes sense to leave it as is, to also show users that they don't _need_ to set a blueprint

closed 2024-03-25 by emilk