rerun-io/rerun #5495: webviewer-react

phanikiranwtc, 2024-03-13:
**Describe the bug**
webviwer-react package is not working, it throws compilation error as below
`ERROR
attempted to open `` in a stopped viewer
    at WebViewer.open (http://localhost:3000/static/js/bundle.js:379616:15)
    at WebViewer.componentDidUpdate (http://localhost:3000/static/js/bundle.js:379373:202)
    at commitLayoutEffectOnFiber (http://localhost:3000/static/js/bundle.js:343379:34)
    at commitLayoutMountEffects_complete (http://localhost:3000/static/js/bundle.js:344520:13)
    at commitLayoutEffects_begin (http://localhost:3000/static/js/bundle.js:344509:11)
    at commitLayoutEffects (http://localhost:3000/static/js/bundle.js:344455:7)
    at commitRootImpl (http://localhost:3000/static/js/bundle.js:346365:9)
    at commitRoot (http://localhost:3000/static/js/bundle.js:346245:9)
    at finishConcurrentRender (http://localhost:3000/static/js/bundle.js:345642:13)
    at performConcurrentWorkOnRoot (http://localhost:3000/static/js/bundle.js:345493:11)`
I see issue https://github.com/rerun-io/rerun/issues/3983 exposed webviewer react package, and i see latest downloads as welll on the package but no issues can be found on the github or else where.

**To Reproduce**
Steps to reproduce the behavior:
1. My application is developed with react 18.2.0 and imported the webviewer package of 0.14.1
2. developed component as hown in the example and i run the application , it throws the above error.

**Expected behavior**
It should have show the webviewer in the component.

**Desktop (please complete the following information):**
 - OS: WIndows 10

**Rerun version**
Rerun webviewer react component package

**Additional context**
i see other package with vite+react, but this is forcing me to migrate my application (atleast) build process to vite, which is complicated for me.. As i am using other ui libraries in my application.

Any help on this to use the react or simple json component would be very help full.

Wumpf, 2024-10-22:
@phanikiranwtc apologies for the huge delay. We're having trouble reproing this, could you share a project/repo that makes it easy to demonstrate the issue?

Wumpf, 2024-11-04:
without additional information I don't see right now how we can act upon this. Please re-open when there's more information, thank you!

closed 2024-11-04 by Wumpf