rerun-io/rerun #5497: Escaped '&' in `?url=` parameter is no longer passed through to rrd request

jleibs, 2024-03-13:
This appears to be a regression as of 0.12.

In order to deal with things likes access tokens it can be helpful to pass through additional url-parameters as part of the rrd uri.

Consider an rrd that needs to be access as: `data.rrd?param1=foo&param2=bar`, escaped as `data.rrd%3Fparam1%3Dfoo%26param2%3Dbar`.

When including these parameters at the end of the url string:

In 0.11, this worked correctly:
https://app.rerun.io/version/0.11.0/?url=https://demo.rerun.io/version/0.11.0/examples/dna/data.rrd%3Fparam1%3Dfoo%26param2%3Dbar

As evidenced in the developer tools:
[image]

However, as of 0.12.0:
https://app.rerun.io/version/0.12.0/?url=https://app.rerun.io/version/0.12.0/examples/dna.rrd%3Fparam1%3Dfoo%26param2%3Dbar

The first parameter is passed through, but not the second:
[image]

jprochazk, 2024-03-13:
This seems to be a bug in egui. The query param parsing done is not percent-decoding things in the right order, it seems: https://github.com/emilk/egui/blob/47fbce665ab64bd2595e960f674427d94b28ec94/crates/eframe/src/web/backend.rs#L123

The entire url is percent-decoded at once which means any extra `&` characters get interpreted as part of the query map. Instead it should first split by `&`, then by `=` and percent-decode each key and value separately.

There's a related issue here which is that any embedded Rerun viewer will assume the `url` parameter is a URL to an RRD file it should load, which is not always true. We should remove usage of `query_map` and instead explicitly pass in the URL into `WebHandle::start`.

emilk, 2024-03-14:
I can fix the egui/eframe code

closed 2024-03-18 by emilk

jprochazk, 2024-03-18:
Opened an issue for the 2nd part here https://github.com/rerun-io/rerun/issues/5555