rerun-io/rerun #5520: C++: Blueprint API parity with python

jleibs, 2024-03-15:
0.15 introduced blueprint support for python

Need to bring the C++ APIs to parity.

samorr, 2025-02-04:
Hey,this feature would be very valuable when thinking of application that needs to preconfigure the viewer to be ready for the end user, or when so many sources of data are logged, that default setting is far from perfect. Do you have any timeline for it?
Thanks!

Wumpf, 2025-02-04:
@samorr unfortunately we don't have a timeline for this yet, so far we haven't been able to prioritize it
Meanwhile the recommendation is to use a Python script to setup the viewer and then use C++(/Rust) for the data logging

teh-cmc, 2025-02-04:
@samorr see also https://rerun.io/docs/howto/visualization/reuse-blueprints

samorr, 2025-02-04:
Oh, so actually there is a way to load existing blueprint in C++, that's much better than separate python script, when in production setup, thanks!