rerun-io/rerun #5521: Rust: Blueprint API parity with python

jleibs, 2024-03-15:
0.15 introduced blueprint support for python

Need to bring the Rust APIs to parity.

adamconkey, 2025-02-06:
Very interested in this one 👍 I was thrilled to see blueprint control was possible from code but was saddened to find it's only in Python currently. I'm building out a visualizer from Rust code so this would be a great help!

Wumpf, 2026-01-07:
From a quick glimpse after https://github.com/rerun-io/rerun/pull/12307 we're missing:
* codegen views rather than implementing each manually
* make it easier to add fitting properties to views (via codegen)
* testing (screenshots? comparision testing with python?)
* get snippets using blueprint to parity with python