rerun-io/rerun #5530: Have Bazel support

the-code-samurai-97, 2024-03-15:
**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**

able add to your bazel repo and then i an include as an dependancy to run
**Describe alternatives you've considered**

**Additional context**

jleibs, 2024-03-15:
Here's an example of how to integrate rerun into a bazel project:

https://github.com/kyle-figure/bazel-minimal-rerun/

the-code-samurai-97, 2024-03-15:
not working```

FAILED: arrow/src/arrow_cpp-stamp/arrow_cpp-configure /private/var/tmp/_bazel_sibinarayananrevathi/7323e08054f8da7553e60a0a337f01b0/sandbox/darwin-sandbox/12/execroot/_main/bazel-out/darwin_arm64-opt/bin/external/rerun/rerun.build_tmpdir/arrow/src/arrow_cpp-stamp/arrow_cpp-configure 
cd /private/var/tmp/_bazel_sibinarayananrevathi/7323e08054f8da7553e60a0a337f01b0/sandbox/darwin-sandbox/12/execroot/_main/bazel-out/darwin_arm64-opt/bin/external/rerun/rerun.build_tmpdir/arrow/src/arrow_cpp-build && /private/var/tmp/_bazel_sibinarayananrevathi/7323e08054f8da7553e60a0a337f01b0/external/cmake-3.23.2-macos-universal/bin/cmake -P /private/var/tmp/_bazel_sibinarayananrevathi/7323e08054f8da7553e60a0a337f01b0/sandbox/darwin-sandbox/12/execroot/_main/bazel-out/darwin_arm64-opt/bin/external/rerun/rerun.build_tmpdir/arrow/src/arrow_cpp-stamp/arrow_cpp-configure-RelWithDebInfo.cmake && /private/var/tmp/_bazel_sibinarayananrevathi/7323e08054f8da7553e60a0a337f01b0/external/cmake-3.23.2-macos-universal/bin/cmake -E touch /private/var/tmp/_bazel_sibinarayananrevathi/7323e08054f8da7553e60a0a337f01b0/sandbox/darwin-sandbox/12/execroot/_main/bazel-out/darwin_arm64-opt/bin/external/rerun/rerun.build_tmpdir/arrow/src/arrow_cpp-stamp/arrow_cpp-configure
CMake Error at /private/var/tmp/_bazel_sibinarayananrevathi/7323e08054f8da7553e60a0a337f01b0/sandbox/darwin-sandbox/12/execroot/_main/bazel-out/darwin_arm64-opt/bin/external/rerun/rerun.build_tmpdir/arrow/src/arrow_cpp-stamp/arrow_cpp-configure-RelWithDebInfo.cmake:49 (message):
  Command failed: 1

   '/private/var/tmp/_bazel_sibinarayananrevathi/7323e08054f8da7553e60a0a337f01b0/external/cmake-3.23.2-macos-universal/bin/cmake' '--preset' 'ninja-release-minimal' '-DARROW_BOOST_USE_SHARED=OFF' '-DARROW_BUILD_SHARED=OFF' '-DARROW_BUILD_STATIC=ON' '-DARROW_CXXFLAGS=' '-DARROW_IPC=OFF' '-DARROW_JEMALLOC=OFF' '-DARROW_MIMALLOC=ON' '-DARROW_USE_ASAN=' '-DARROW_USE_TSAN=OFF' '-DARROW_USE_UBSAN=OFF' '-DBOOST_SOURCE=BUNDLED' '-DCMAKE_INSTALL_PREFIX=/private/var/tmp/_bazel_sibinarayananrevathi/7323e08054f8da7553e60a0a337f01b0/sandbox/darwin-sandbox/12/execroot/_main/bazel-out/darwin_arm64-opt/bin/external/rerun/rerun.build_tmpdir/arrow' '-Dxsimd_SOURCE=BUNDLED' '-DBOOST_SOURCE=BUNDLED' '-DARROW_BOOST_USE_SHARED=OFF' '-GNinja' '/private/var/tmp/_bazel_sibinarayananrevathi/7323e08054f8da7553e60a0a337f01b0/sandbox/darwin-sandbox/12/execroot/_main/bazel-out/darwin_arm64-opt/bin/external/rerun/rerun.build_tmpdir/arrow/src/arrow_cpp/cpp'

  See also

    /private/var/tmp/_bazel_sibinarayananrevathi/7323e08054f8da7553e60a0a337f01b0/sandbox/darwin-sandbox/12/execroot/_main/bazel-out/darwin_arm64-opt/bin/external/rerun/rerun.build_tmpdir/arrow/src/arrow_cpp-stamp/arrow_cpp-configure-*.log

ninja: build stopped: subcommand failed.
_____ END BUILD LOGS _____
rules_foreign_cc: Build wrapper script location: bazel-out/darwin_arm64-opt/bin/external/rerun/rerun_foreign_cc/wrapper_build_script.sh
rules_foreign_cc: Build script location: bazel-out/darwin_arm64-opt/bin/external/rerun/rerun_foreign_cc/build_script.sh
rules_foreign_cc: Build log location: bazel-out/darwin_arm64-opt/bin/external/rerun/rerun_foreign_cc/CMake.log

Target //src/examples/rerun_example:rerun_example failed to build
Use --verbose_failures to see the command lines of failed build steps.
(23:10:15) ERROR: /Users/sibinarayananrevathi/dev/learning_workspace/src/examples/rerun_example/BUILD:1:10 Linking src/examples/rerun_example/rerun_example failed: (Exit 1): bash failed: error executing CcCmakeMakeRule command (from target @@rerun//:rerun) /bin/bash -c bazel-out/darwin_arm64-opt/bin/external/rerun/rerun_foreign_cc/wrapper_build_script.sh
```

Wumpf, 2024-03-15:
Hard to read anything into that error, but I'm closing this since we don't maintain Bazel support ourselves and don't really have the capacity to provide support for it right now. For more details see

* #4579

closed 2024-03-15 by Wumpf