rerun-io/rerun #5532: I can't compile on my Mac

the-code-samurai-97, 2024-03-15:
make ..
Compiler: AppleClang 15.0.0.15000100 (/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++)
-- Rerun SDK install version: 0.15.0
CMake Error at tests/CMakeLists.txt:22 (rerun_strict_warning_settings):
  Unknown CMake command "rerun_strict_warning_settings".
**Describe the bug**

**To Reproduce**
Steps to reproduce the behavior:
1. 
2. 
3. 
4. 
make ..
Compiler: AppleClang 15.0.0.15000100 (/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++)
-- Rerun SDK install version: 0.15.0
CMake Error at tests/CMakeLists.txt:22 (rerun_strict_warning_settings):
  Unknown CMake command "rerun_strict_warning_settings".
**Expected behavior**

**Screenshots**

**Backtrace**

**Desktop (please complete the following information):**
 - OS: 

**Rerun version**

**Additional context**

jleibs, 2024-03-15:
Please provide additional context on the steps followed to create your environment.  None of our instructions involve running "make .." so I'm unsure what you're trying to do.

I would recommend to start by following our C++ getting started guide: https://www.rerun.io/docs/getting-started/cpp

the-code-samurai-97, 2024-03-15:
@jleibs sorry for an messy bug update , i have updated

the-code-samurai-97, 2024-03-15:
Actual Full output
```
-- The CXX compiler identification is AppleClang 15.0.0.15000100
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Compiler: AppleClang 15.0.0.15000100 (/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++)
-- Rerun SDK install version: 0.15.0
-- Performing Test HAVE_FLAG__ffile_prefix_map__Users_sibinarayananrevathi_dev_rerun_rerun_cpp_build__deps_catch2_src__
-- Performing Test HAVE_FLAG__ffile_prefix_map__Users_sibinarayananrevathi_dev_rerun_rerun_cpp_build__deps_catch2_src__ - Success
CMake Error at tests/CMakeLists.txt:22 (rerun_strict_warning_settings):
  Unknown CMake command "rerun_strict_warning_settings".

-- Configuring incomplete, errors occurred!
```

Wumpf, 2024-03-15:
That error looks like you first navigated to `rerun_cpp/build` which is why it can't pick up the top level CMake file and thus doesn't find `rerun_strict_warning_settings` - this cleanly repros for me if I do it like that.
If you want to build Rerun from the repository you need to execute CMake with the top level directory as working directory. However, I highly recommend using pixi instead for working within the rerun repository as this takes care of most dependencies.

Generally, building Rerun directly from the repository is only relevant if you want to contribute to Rerun. If you instead want to use Rerun yourself or try out some of the examples you should use a Rerun C++ sdk bundle that's provided with each release. To learn more about that check the page @jleibs already linked (https://www.rerun.io/docs/getting-started/cpp)

closed 2024-03-15 by Wumpf