rerun-io/rerun #5541: Built in data-loader support for common archieve formats: `.tar`, `.zip`, etc.

jleibs, 2024-03-15:
We already have dataloader support for directories.

Would be very handy if we just did the right thing if someone puts a `.rbl` and several `.rrd` together in a Zip and then drag-and-drops it on the viewer.

This could be a nice convenience alternative to inlining blueprints in rrd files.

teh-cmc, 2024-03-15:
- Duplicate of https://github.com/rerun-io/rerun/issues/5052
  > It's very annoying to have to download zip archives that people host somewhere just so we can manually unzip em just to be able to load them.
  > 
  > When loading a file via http, this can probably just rely on the http client's auto-dezip support.

I'll close the other since this one is a bit more generalist.

Wumpf, 2024-04-16:
Related to
* #4224