rerun-io/rerun #5583: "Reset blueprint" button should reset to the blueprint in the recording (if any)

emilk, 2024-03-19:
The "Reset blueprint" button currently clears the blueprint, going to the heuristic definition.

Instead, if there was a blueprint bundled with the recording, we should reset to that.

"Reset to heuristics" is arguably a much less useful feature, and could perhaps be hidden in a conext menu on the reset button.

---

To implement this we need to keep track of which blueprint came with which recording. We can do so via the data sources (if they came from the same source, they belong together).

emilk, 2024-03-25:
We should probably save all the blueprints on shutdown, not just the active ones.

jleibs, 2024-03-27:
Closing based on the work done in: https://github.com/rerun-io/rerun/pull/5678

Note, this doesn't strictly implement the feature as described. Reset will reset to the current default blueprint (which may or may not be the one from the given recording). I believe this is close enough given our discussions on downsides of the "always use the recording-blueprint". We can open a new issue if necessary.

closed 2024-03-27 by jleibs