rerun-io/rerun #5589: The viewer wastes too much vertical space

teh-cmc, 2024-03-20:
In general, I find that:
- The top bar is way too tall, with an information density dangerously approaching zero.
- The tab bars are also way too tall.
- The collapsed time panel is, once again, also way too tall.
- And of course the collapsed time panel should never turn into two things like it does in evidence 1 (first time I see that one to be fair).

As someone with a long-love for Vim in part because of how much it respects the user's screen estate, this is a really painful one...

Of course all of these get worse on small screens, i.e. iframes and mobiles.

My proposal would be to A) make all these bars much thinner and maybe B) make them responsive based on the current height of the app, with some fixed minimum and maximum bounds.

Evidence 1:
[image]

Evidence 2:
[image]

Evidence 3:
[image]

teh-cmc, 2024-03-20:
Extra datapoint, a 3x3 grid in vim:
[image]

Vim famously strips away all tab bars within the grid itself to save on screen space and information density, with the assumption that you already know what is what just by the looking at the contents anyway.
In doubt, more info about the currently selected cell can be found in the one single status bar at the bottom.

abey79, 2024-03-20:
#### Title bar

Specifically on mac, the title bar takes _exactly_ the right vertical space in windowed mode, which is the same as any other (standard) window (though they tend to becoming increasingly rare). As soon as it goes full screen, then it's back to using too much space as per your screenshot.

[image]

#### Tab bar

For the tab bar, I agree, but it's not very many pixels. They can't be thiner than e.g. the `ListItem`. Exhibit:

[image]

With the advent of blueprint, tree drag and drop, and use on the web, I think we should also consider a mode where they are hidden altogether.

#### Time panel

Agree we can shave some pixel. Agree the responsive mobile mode is not ideal (unsure how to fix). A mode where it's hidden altogether is needed as well.

teh-cmc, 2024-03-20:
> With the advent of blueprint, tree drag and drop, and use on the web, I think we should also consider a mode where they are hidden altogether.

Yeah I'm more and more convinced by this.

jleibs, 2024-03-20:
> Vim famously strips away all tab bars within the grid itself to save on screen space and information density, with the assumption that you already know what is what just by the looking at the contents anyway.

I disagree on this one. This may be true for some power users/layouts. But I think having the ability to show clear titles for the space views (and containers!) is crucial for new users to understand what's going on and build the link between the blueprint tree and the layout itself.

Also the assertion that you know what's there by the contents only holds when things are working well and are visually distinct. Users spend lots and lots of time with space views not showing what they're supposed to. Even if configured properly, a key use case of rerun is helping users understand when their own code is not producing the anticipated results. 

Furthermore, Vim is famously unfriendly to new users -- I don't think this is the right choice for us to emulate.

emilk, 2024-08-07:
A good example why this matters is the https://www.yaak.ai/Research web site:

[image]