rerun-io/rerun #5596: Show software rasterizer warning in title bar

Wumpf, 2024-03-20:
It's easy to accidentally run on Linux into a situation where Rerun is rendered using the software rasterizer, quickly leading to abyssmal performance. We log a warning on commandline but it's easy to miss.

Simply style it like our "Debug Build" warning and have it link to the troubleshoot guide.

(The troubleshoot guide is not _always_ helping out with this right now, but that's a separate issue we (and wgpu) have to work on)

emilk, 2024-03-20:
Even better would be to show _all_ logged warnings and errors in an easily visible way, i.e. as a "notification" bar in the top bar. We used to have an issue for this, but now I can't find it.

Wumpf, 2024-03-20:
agreed, our log notification system needs an overhaul. But let's stick with the simple thing here :)

Wumpf, 2024-03-20:
Separate issue for that here:
* #5596

closed 2024-03-25 by emilk