rerun-io/rerun #5639: Privacy policy

gokcergun, 2024-03-21:
Hi, 
Sorry for writing my question here, I tried to connect Discord server provided, but I'm getting an 'invitation cannot be accepted' error. 

I'd like to double-check with you the privacy policy that you published. It is only relevant for the web usage, but not for the local usage (installing rerun using pip). And the user doesn't send any information/data during the local usage. Is it true? 

Thank you.

jleibs, 2024-03-21:
We do collection some anonymous usage statics from the rerun native application, including installations via `pip`.

You can find the details here: https://github.com/rerun-io/rerun/blob/main/crates/re_analytics/README.md

As mentioned there, you can opt out of local analytics by running the command `rerun analytics disable`

jleibs, 2024-03-21:
> I tried to connect Discord server provided

Also, which link did you follow for the discord invite? The link from https://www.rerun.io/ still appears to be working.

gokcergun, 2024-03-22:
Thanks a lot for the answer. My main concern was related to the privacy of the images that I'm logging to the Rerun, but it seems that you don't collect any logged data in anyway. So, the provided images are neither processed nor transferred, isn't it?

When I create a new Issue - an automatic message appears:
If you are asking a question, use [the Rerun Discord server](https://discord.gg/PXtCgFBSmH) instead.-
I was using this link.

jleibs, 2024-03-22:
Correct -- we are sensitive to user privacy needs in that regard, and our analytics system will never collect any logged user data.

However, to clarify, the SDK and Viewer do run as separate processes and are capable of operating over the network.

When you call the `.connect()` function in the SDK it initiates a TCP connection to the Viewer process. By default it will only connect to local-host, but you can configure it to use a different host.

So long as you only run the Viewer and SDK locally, your data will stay on your own network.

gokcergun, 2024-03-22:
Thank you for detailed explanation. 

By the way, I tried the discord link directly from the homepage. I'm still getting the error message. 
[image]

Discord support: https://support.discord.com/hc/en-us/articles/360001556852-Invalid-Invite-Links

I checked the discord support, but still not sure about the reason.  I tried again after signed-up discord, but did not help.

Wumpf, 2024-03-25:
Checked on the link and couldn't repro, maybe it was intermittent 🤷

closed 2024-03-25 by Wumpf