rerun-io/rerun #5649: Layout code panics

teh-cmc, 2024-03-22:
Didn't do anything particular... I'll come up with a repro later hopefully.

```
thread 'main' panicked at 'assertion failed: child_size.x >= 0.0 && child_size.y >= 0.0'
egui/src/layout.rs:600
stack backtrace:
   6: core::panicking::panic_fmt
             at core/src/panicking.rs:72:14
   7: core::panicking::panic
             at core/src/panicking.rs:127:5
   8: egui::layout::Layout::next_frame_ignore_wrap
             at egui/src/layout.rs:600:9
   9: egui::layout::Layout::next_widget_space_ignore_wrap_justify
             at egui/src/layout.rs:658:21
  10: egui::placer::Placer::next_widget_space_ignore_wrap_justify
             at egui/src/placer.rs:214:9
      egui::placer::Placer::set_max_width
             at egui/src/placer.rs:221:20
      egui::ui::Ui::set_max_width
             at egui/src/ui.rs:512:9
  11: re_viewer::ui::welcome_screen::welcome_section::onboarding_content_ui::{{closure}}::{{closure}}::{{closure}}::{{closure}}
             at re_viewer/src/ui/welcome_screen/welcome_section.rs:244:29
      core::ops::function::FnOnce::call_once{{vtable.shim}}
             at core/src/ops/function.rs:250:5
  12: <alloc::boxed::Box<F,A> as core::ops::function::FnOnce<Args>>::call_once
             at alloc/src/boxed.rs:2007:9
  13: egui::ui::Ui::with_layout_dyn
             at egui/src/ui.rs:2068:21
  14: egui::ui::Ui::vertical
             at egui/src/ui.rs:1996:9
      re_viewer::ui::welcome_screen::welcome_section::onboarding_content_ui::{{closure}}::{{closure}}::{{closure}}
             at re_viewer/src/ui/welcome_screen/welcome_section.rs:242:25
      core::ops::function::FnOnce::call_once{{vtable.shim}}
             at core/src/ops/function.rs:250:5
  15: <alloc::boxed::Box<F,A> as core::ops::function::FnOnce<Args>>::call_once
             at alloc/src/boxed.rs:2007:9
  16: egui::grid::Grid::show_dyn::{{closure}}::{{closure}}
             at egui/src/grid.rs:450:25
      core::ops::function::FnOnce::call_once{{vtable.shim}}
             at core/src/ops/function.rs:250:5
  17: <alloc::boxed::Box<F,A> as core::ops::function::FnOnce<Args>>::call_once
             at alloc/src/boxed.rs:2007:9
      egui::ui::Ui::allocate_ui_with_layout_dyn
             at egui/src/ui.rs:946:19
  18: egui::ui::Ui::horizontal_with_main_wrap_dyn
             at egui/src/ui.rs:1978:9
  19: egui::ui::Ui::horizontal
             at egui/src/ui.rs:1906:9
      egui::grid::Grid::show_dyn::{{closure}}
             at egui/src/grid.rs:430:16
      egui::ui::Ui::allocate_ui_at_rect
             at egui/src/ui.rs:967:19
  20: egui::grid::Grid::show_dyn
             at egui/src/grid.rs:428:9
  21: egui::grid::Grid::show
             at egui/src/grid.rs:395:9
      re_viewer::ui::welcome_screen::welcome_section::onboarding_content_ui::{{closure}}::{{closure}}
             at re_viewer/src/ui/welcome_screen/welcome_section.rs:225:13
      core::ops::function::FnOnce::call_once{{vtable.shim}}
             at core/src/ops/function.rs:250:5
  22: <alloc::boxed::Box<F,A> as core::ops::function::FnOnce<Args>>::call_once
             at alloc/src/boxed.rs:2007:9
  23: egui::ui::Ui::with_layout_dyn
             at egui/src/ui.rs:2068:21
  24: egui::ui::Ui::vertical
             at egui/src/ui.rs:1996:9
      re_viewer::ui::welcome_screen::welcome_section::onboarding_content_ui::{{closure}}
             at re_viewer/src/ui/welcome_screen/welcome_section.rs:217:9
      core::ops::function::FnOnce::call_once{{vtable.shim}}
             at core/src/ops/function.rs:250:5
[... 136 lines cut ...]
      std::thread::local::LocalKey<T>::with
             at std/src/thread/local.rs:246:9
      eframe::native::run::with_event_loop
             at eframe/src/native/run.rs:48:16
      eframe::native::run::run_wgpu
             at eframe/src/native/run.rs:421:16
      eframe::run_native
             at eframe/src/lib.rs:268:13
  56: extend_viewer_ui::main::{{closure}}
             at extend_viewer_ui/src/main.rs:43:5
      tokio::runtime::park::CachedParkThread::block_on::{{closure}}
             at tokio-1.28.1/src/runtime/park.rs:283:63
      tokio::runtime::coop::with_budget
             at tokio-1.28.1/src/runtime/coop.rs:107:5
      tokio::runtime::coop::budget
             at tokio-1.28.1/src/runtime/coop.rs:73:5
      tokio::runtime::park::CachedParkThread::block_on
             at tokio-1.28.1/src/runtime/park.rs:283:31
  57: tokio::runtime::context::BlockingRegionGuard::block_on
             at tokio-1.28.1/src/runtime/context.rs:315:13
      tokio::runtime::scheduler::multi_thread::MultiThread::block_on
             at tokio-1.28.1/src/runtime/scheduler/multi_thread/mod.rs:68:14
  58: tokio::runtime::runtime::Runtime::block_on
             at tokio-1.28.1/src/runtime/runtime.rs:304:45
```

emilk, 2024-03-22:
It’s hard to tell, but this is actually a assert that is only on debug builds. Still, it indicates that there is some UI that is smaller than zero, which probably means a layout bug

Wumpf, 2024-03-22:
looking at the code a bit that figures out the min width that the assert is stumbling on, isn't bounded despite a bound being defined - pushing a speculative fix that will make clear what I mean

PR #5650 (2024-03-22): Fix layouting issue on welcome screen for narrow window, triggering debug assertion
### What

* Fixes #5649

~This is a speculative fix, didn't have a repro.~ From the callstack we know that a certain `ui.set_max_width(column_width - 8.0);` might have gotten a too small width. Looking at how `column_width` is produced this may very well happen if for some reason there's very little `available_width`.

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and the [Code of Conduct](https://github.com/rerun-io/rerun/blob/main [... cut]
commit e9e90254 (2024-03-22): Fix layouting issue on welcome screen for narrow window, triggering debug assertion (#5650)

### What

* Fixes #5649

~This is a speculative fix, didn't have a repro.~ From the callstack we
know that a certain `ui.set_max_width(column_width - 8.0);` might have
gotten a too small width. Looking at how `column_width` is produced this
may very well happen if for some reason there's very little
`available_width`.

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/ [... cut]
closed 2024-03-22 by Wumpf