rerun-io/rerun #5665: New categorization system for examples

nikolausWest, 2024-03-25:
Replaces
- https://github.com/rerun-io/rerun/issues/5454

In-progress PR:
- https://github.com/rerun-io/rerun/pull/5599

We've decided to redo how examples are categorized. In stead of the current very-arbitrary categories, we'll opt for categorizing (loosely) along use-cases (see the PR to get a sense). The categories can be overlapping if useful and can include things like "Spatial computing and XR" and "Robotics", but potentially also things like "Integrations".

The presentation of these categories should be on a single page under rerun.io/examples as:
```
<Category title>
<Category description (1-2 lines max)>
<example card with thumbnail, title, and tags>, <example card with thumbnail, title, and tags>, <example card with thumbnail, title, and tags>, ...

<Category title>
<Category description (1-2 lines max)>
<example card with thumbnail, title, and tags>, <example card with thumbnail, title, and tags>, <example card with thumbnail, title, and tags>, ...

<Category title>
<Category description (1-2 lines max)>
<example card with thumbnail, title, and tags>, <example card with thumbnail, title, and tags>, <example card with thumbnail, title, and tags>, ...

...
```

emilk, 2024-03-27:
Should each example be in just one category? If so, which one do we pick in case of overlap?

emilk, 2024-04-03:
This could use a Figma link or a screenshot of what we're aiming for

closed 2024-04-04 by abey79