rerun-io/rerun #5666: Add all external examples to the example manifest

nikolausWest, 2024-03-25:
There are several useful examples that we keep in separate repositories that aren't currently easy to find under rerun.io/examples. We should add these to the example manifest to make the easier to find for our users. This includes, any ROS bridges or interesting dataloaders.

Use [this figma page](https://www.figma.com/file/U85f12k8Tw2ArD4N61cdtQ/Marketing-material?type=design&node-id=2289-2063&mode=design&t=ethbZToeEgZeUcf5-0) to format the thumbnail.

Related to:
- https://github.com/rerun-io/rerun/issues/5665

emilk, 2024-03-27:
We could do it in https://github.com/rerun-io/rerun/pull/5599 to avoid merge conflicts

emilk, 2024-03-27:
We should maybe add a lint so that we don't forget adding examples to the manifest