rerun-io/rerun #5670: "Which is the active blueprint?" tracking issues

emilk, 2024-03-25:
Common issues:

* #5583

jleibs, 2024-03-25:
Proposal for 0.15 as simplest thing that works:
 - Two blueprints per app-id:  one "active", one "stored"
 - Incoming blueprint always updates "stored" and only makes the stored blueprint active if there is no existing blueprint
 - https://github.com/rerun-io/rerun/issues/5583 always resets to the stored blueprint

Reasoning:
 - User edits are never lost unless they explicitly click the reset button
 - Clicking reset is a single click in a blueprint editing workflow
 - Other use-cases can be supported (though hackily) by using modified app-ids

jleibs, 2024-03-25:
Optional improvement:
 - Advanced UI menu on recording selection to activate a **specific** blueprint instead of only ever allowing user to activate he last one.

Wumpf, 2025-07-07:
I think we covered this well enough by now. Or at least we moved forward such that the language in here no longer applies 🤔