rerun-io/rerun #5682: Improve contrast of our Design

emilk, 2024-03-26:
Currently our design has very low contrast between elements. Here is a screenshot from https://www.figma.com/file/3WX8yUqa8ECmhLwV6NIF3a/Rerun-Design-System?type=design&node-id=1252%3A15687&mode=design&t=Xc27rEOvAdS7F0XZ-1

[image]

---

This looks _great_ if you
* have a good screen
* have good eyes
* are in a dark room

However, if you are using a bad screen in a light room, the contrast between these elements is way too low.
We either need to make the contrast higher, or add border strokes or similar.

Wumpf, 2024-03-26:
Related to
* #1839

nikolausWest, 2024-03-26:
TBH, the screenshot above still has better contrast than our actual app:

[image]

Wumpf, 2024-03-26:
not the bit where the context menu overlaps the background:
Actual:
[image]

Figma:
[image]