rerun-io/rerun #5694: Opening a blueprint appears to do nothing if you have no rrd files

jleibs, 2024-03-26:
We are dependent upon the existance of Recordings to manage whether an app shows up in the recordings.

However, if I open a blueprint, I expect to see the app show up, regardless of whether I have any recordings.

emilk, 2024-03-26:
In other words, you want to see all app ids in the blueprint panel, regarless if they have any recordings or not?
What if you open a recording and close it again, do we still show the app id? Or do we throw out the blueprint once we close the last recording using it?

jleibs, 2024-03-27:
If we make app an explicit, selectable concept rather than an implicit grouping of recordings I think it warrants its own close button somewhere.

emilk, 2024-04-02:
Suggestion:
* Switch to the app id when opening a blueprint (even if it has no recording)
* Show all known App Ids in the Blueprint panel that has at least one blueprint OR recording
* Close all blueprints of an App Id when closing the last recording of that app ID

PR #5766 (2024-04-04): Show all loaded applications in recordings panel
* Closes #5672
* Closes #5694 

### What
* Show all loaded applications in the recordings panel
  * Even those with no recordings
  * Even welcome screen
* Close application when last recording of that app is closed
* Add button to close all recordings in an application
* Make applications selectable
* When loading a blueprint, switch to its application

Being able to quickly select the welcome screen is super useful while exploring examples, and for testing.
It is also the natural place to return to to find the links to the docs, and maybe more [... cut]
commit df4c1300 (2024-04-04): Show all loaded applications in recordings panel (#5766)

* Closes #5672
* Closes #5694 

### What
* Show all loaded applications in the recordings panel
  * Even those with no recordings
  * Even welcome screen
* Close application when last recording of that app is closed
* Add button to close all recordings in an application
* Make applications selectable
* When loading a blueprint, switch to its application

Being able to quickly select the welcome screen is super useful while
exploring examples, and for testing.
It is also the natural place to return to to find the links to the docs,
and m [... cut]
closed 2024-04-04 by emilk