rerun-io/rerun #5695: If launching rerun with a blueprint from the command-line there's no way to send data to the app

jleibs, 2024-03-26:
We don't appear to start the TCPServer if we pass a local file to rerun on start.

Although we've always had this issue when loading `.rrd` files from the CLI, the fact that Blueprint files and RRD files combine makes this particularly annoying.

Consider the following workflow:

I've saved a blueprint.

Now I want to run:
```
rerun my_app.rbl
```
to restore it (related issue: https://github.com/rerun-io/rerun/issues/5694)

However, because there is no TCPServer running, I can't run an app that connects to it.

teh-cmc, 2024-03-27:
On a related note, I've wanted to be able to use the CLI as a data forwarder tool for a long time, e.g.: `rerun my_data.rrd --connect` (or, in this instance, `rerun my_blueprint.rbl --connect`):
- https://github.com/rerun-io/rerun/issues/2044

emilk, 2024-04-02:
The easy solution is to always spawn a TCP server when we launch the `rerun` binary (unless `--no-server` or whatever)