rerun-io/rerun #5697: Panic upon exiting program if a recording is stored in a global variable

02alexander, 2024-03-26:
**Describe the bug**
If I store a recording as a global variable the program will panic upon exiting the program. 
The following code 
```cpp
#include <rerun.hpp>

rerun::RecordingStream rec("test");

int main() {
    rec.spawn().exit_on_failure();
}

```
outputs the following when run:
```
$ RUST_BACKTRACE=1 ./panic_example 
[2024-03-26T19:17:47Z INFO  re_sdk_comms::server] Hosting a SDK server over TCP at 0.0.0.0:9876. Connect with the Rerun logging SDK.
[2024-03-26T19:17:47Z WARN  winit::platform_impl::platform::x11::xdisplay] error setting XSETTINGS; Xft options won't reload automatically
[2024-03-26T19:17:47Z INFO  winit::platform_impl::platform::x11::window] Guessed window scale factor: 1
[2024-03-26T19:17:47Z INFO  egui_wgpu] There were 3 available wgpu adapters: {backend: Vulkan, device_type: DiscreteGpu, name: "NVIDIA GeForce RTX 3060", driver: "NVIDIA", driver_info: "545.29.06", vendor: 0x10DE, device: 0x2487}, {backend: Vulkan, device_type: IntegratedGpu, name: "AMD Radeon Graphics (RADV RAPHAEL_MENDOCINO)", driver: "radv", driver_info: "Mesa 23.3.5", vendor: 0x1002, device: 0x164E}, {backend: Vulkan, device_type: Cpu, name: "llvmpipe (LLVM 16.0.6, 256 bits)", driver: "llvmpipe", driver_info: "Mesa 23.3.5 (LLVM 16.0.6)", vendor: 0x10005}
[2024-03-26T19:17:48Z INFO  tracing::span] perform;
[2024-03-26T19:17:48Z INFO  zbus::handshake] write_command; command=Auth(Some(External), Some([49, 48, 48, 48]))
[2024-03-26T19:17:48Z INFO  tracing::span] read_command;
[2024-03-26T19:17:48Z INFO  zbus::handshake] write_command; command=NegotiateUnixFD
[2024-03-26T19:17:48Z INFO  tracing::span] read_command;
[2024-03-26T19:17:48Z INFO  zbus::handshake] write_command; command=Begin
[2024-03-26T19:17:48Z INFO  tracing::span] socket reader;
[2024-03-26T19:17:48Z INFO  re_sdk_comms::server] New SDK client connected: 127.0.0.1:51458
[2024-03-26T19:17:48Z INFO  re_sdk_comms::server] New SDK client connected: 127.0.0.1:51470
thread '<unnamed>' panicked at library/std/src/thread/mod.rs:711:35:
use of std::thread::current() is not possible after the thread's local data has been destroyed
stack backtrace:
   0: rust_begin_unwind
             at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/std/src/panicking.rs:597:5
   1: core::panicking::panic_fmt
             at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/core/src/panicking.rs:72:14
   2: core::panicking::panic_display
             at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/core/src/panicking.rs:168:5
   3: core::panicking::panic_str
             at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/core/src/panicking.rs:152:5
   4: core::option::expect_failed
             at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/core/src/option.rs:1988:5
   5: core::option::Option<T>::expect
             at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/core/src/option.rs:898:21
   6: std::thread::current
             at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/std/src/thread/mod.rs:711:35
   7: crossbeam_channel::context::Context::new
   8: crossbeam_channel::flavors::zero::Channel<T>::recv
   9: crossbeam_channel::channel::Receiver<T>::recv
  10: re_log_types::data_table_batcher::DataTableBatcherInner::flush_blocking
  11: core::option::Option<T>::map
  12: re_sdk::recording_stream::RecordingStream::set_sink
  13: rr_recording_stream_free
  14: _ZN5rerun15RecordingStreamD1Ev
  15: __run_exit_handlers
  16: exit
  17: __libc_start_call_main
  18: __libc_start_main_impl
  19: _start
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
[1]    317655 IOT instruction (core dumped)  RUST_BACKTRACE=1 ./panic_example
```

**Desktop (please complete the following information):**
 - OS: NixOS 

**Rerun version**
rerun-cli 0.15.0-alpha.3 [rustc 1.74.0 (79e9716c9 2023-11-13), LLVM 17.0.4] x86_64-unknown-linux-gnu

**Additional context**
The cmake file fetches https://build.rerun.io/commit/43cbc67/rerun_cpp_sdk.zip

Wumpf, 2024-08-02:
Can't repro this on Windows, will try on Mac soon.

Hypothesis: This is a problem of cleanup order between Rust and C++
No conclusive evidence for this yet, but we know that if we destroy the object before exiting main (put it into a unique_ptr) then we don't crash.

emilk, 2024-08-05:
This looks like a duplicate of
* https://github.com/rerun-io/rerun/issues/5260

closed 2024-08-05 by emilk