rerun-io/rerun #5721: Increase thickness of lines drawn between keypoint connections

palol, 2024-03-28:
**Is your feature request related to a problem? Please describe.**
I have not been able to find an example that demonstrates drawing thicker/thinner keypoint skeletons using the `AnnotationContext` class. Currently, it seems like the only solution is to instead draw the skeleton using `LineStrips`. 

**Describe the solution you'd like**
Would it be possible to expose another variable like `radii` to the `AnnotationContext` class, which would enable adjusting the thickness of the drawn keypoint skeleton. 

**Describe alternatives you've considered**
Alternatively, including a code example using `LineStrips` to draw a custom skeleton for a set of keypoints. 

**Additional context**
Example change
```python
rr.log(
        "/",
        rr.AnnotationContext(
            rr.ClassDescription(
                info=rr.AnnotationInfo(id=0, label="Person"),
                keypoint_annotations=[rr.AnnotationInfo(id=lm.value, label=lm.name) for lm in mp_pose.PoseLandmark],
                keypoint_connections=mp_pose.POSE_CONNECTIONS,
                **radii=0.25**
            )
        ),
        timeless=True,
    )
```

jleibs, 2024-03-28:
Makes sense.

In the short term you can actually influence this behavior by selecting the space view and then modifying the "Default size" of line radius.

Exa:
[image]

However, I agree we should be able to control this explicitly via the API.

palol, 2024-03-29:
> Makes sense.
> 
> In the short term you can actually influence this behavior by selecting the space view and then modifying the "Default size" of line radius.
> 
> However, I agree we should be able to control this explicitly via the API.

Thank you! I new there had to be a global setting somewhere in the viewer.

Wumpf, 2024-04-23:
As pointed out in
* https://github.com/rerun-io/rerun/issues/6064

this should naturally also be per class, not just on the entire context

Wumpf, 2024-10-22:
looks like we regressed further on this: now the workaround doesn't work either anymore