rerun-io/rerun #5752: Problem accessing example manifest results in bad welcome page

jleibs, 2024-04-02:
I tested out `-rc.1` before the manifest had full published and was welcomed to this screen:

[image]

Even if we can't reach/process the manifest we should probably have a fallback welcome screen that is more user-friendly.

emilk, 2024-04-03:
This should be a fairly easy fix: just log the error instead of showing it, and then only show this portion to the user:

[image]

emilk, 2024-04-03:
The first part of the welcome page should also show up right away. Right now it is waiting for the example manifest to download, which is unnecessary.

closed 2024-04-03 by emilk