rerun-io/rerun #5783: Properly format tags in the welcome screen's example card

abey79, 2024-04-04:
In app vs web, notice the difference in tag formatting:

[image]
[image]

We need to capitalise and remove `-`

emilk, 2024-04-04:
Blocked by https://github.com/rerun-io/rerun/pull/5779 (will lead to too much merge conflicts)

abey79, 2024-04-04:
My idea was to do that in the UI code rather than actually changing all tags. That's apparently how it's done for the landing page.

abey79, 2024-04-04:
Not convinced this is 0.15 blocking, though—just added to the milestone for visibility.

emilk, 2024-04-04:
We decided on https://rerunio.slack.com/archives/C04UBTMMGJK/p1712250738168869 to not do any conversions, and instead use `Human case` directly in the frontmatter.

PR #5802 (2024-04-05): Change all example tags to "Human case"
### What

- Closes #5783

@jprochazk is there some tag formatting code on landing that you could remove?

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and the [Code of Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested the web demo (if applicable):
  * Using newly built examples: [rerun.io/viewer](https://rerun.io/viewer/pr/5802)
  * Using examples from latest `main` build: [rer [... cut]
commit 695e195a (2024-04-05): Change all example tags to "Human case" (#5802)

### What

- Closes #5783

@jprochazk is there some tag formatting code on landing that you could
remove?

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested the web demo (if applicable):
* Using newly built examples:
[rerun.io/viewer](https://rerun.io/viewer/pr/5802)
* Using examples from latest `main` build: [... cut]
closed 2024-04-05 by abey79