rerun-io/rerun #5799: Reference all performance bottlenecks highlighted by "air traffic" example

teh-cmc, 2024-04-05:
The [recently added air traffic example](https://github.com/rerun-io/rerun/pull/5449) highlights _a lot_ of performance bottlenecks in every single corner of the app (and probably SDK too now that I think of it...).

All these bottlenecks should have a dedicated issue explaining what's going on so we can investigate them at some point.

The first thing to try out is the cleaned up dataset, which demonstrate a lot of performance issues with normal looking data:
```
python examples/python/air_traffic_data/main.py --dataset 2h
```

and then move on to the raw dataset, which adds another layer of issues specific to ill-formed data (which ironically is more likely to be what's in the wild):
```
python examples/python/air_traffic_data/main.py --dataset 2h --raw
```

abey79, 2024-04-05:
related: https://github.com/rerun-io/rerun/issues/5812

teh-cmc, 2024-07-09:
Let's see how that behaves once `0.18` is ready to go.

emilk, 2024-08-08:
Likely:
* https://github.com/rerun-io/rerun/issues/7115

emilk, 2024-08-08:
Around a third of the frame-time is just asking entities wether or not they have a specific component or data on some timeline. If that was cached, it would be a huge win

[image]

abey79, 2024-08-08:
0.18 significantly increase frame time on my computer (3d view latest at and one plot with all data)

top 0.17, bottom main as of today.
[image]

Wumpf, 2024-12-13:
We have a decent understanding of this by now
* https://github.com/rerun-io/rerun/issues/8233

closed 2024-12-13 by Wumpf