rerun-io/rerun #5822: Embedded video doesn't work in doc-preview

jleibs, 2024-04-05:
For some reason in the doc previews, videos don't appear to work.

See: https://landing-8zs0bvf65-rerun.vercel.app/docs/getting-started/data-in/cpp

[image]

jprochazk, 2024-04-06:
Seems like the markdown parser we use doesn't parse HTML properly. It doesn't ignore the whitespace inbetween elements, or maybe it doesn't check the void element list before treating an element as one. In this case, the offending source code is:

```html
<video width="100%" autoplay loop muted controls>

<source src="https://static.rerun.io/c4c4ef1e4a1b25002da7c44d4316b0e07ae8d6ed_logging_data1_result.webm" type="video/webm" />
</video>
```

It seems to parse the `video` element as a void element, and the `source` + closing `video` tag are then ignored, because `source` does nothing on its own, and a closing tag in an invalid position is discarded.

jprochazk, 2024-04-06:
Here's the workaround: https://github.com/rerun-io/rerun/pull/5826

Switching to a different markdown parser would be significantly more work right now

closed 2024-04-07 by nikolausWest