rerun-io/rerun #5823: Advanced camera rotation controls

jleibs, 2024-04-06:
The default orbit camera maintains an up-axis and subsequently gets "stuck" at the extreme rotations where the definition of up would become inverted.

While the behavior of maintaining up can be helpful to prevent users from getting disoriented with 3D content, there are cases where advanced users would like to have arbitrary control of all 3 axes without these limitations.

Being able to enable this as an camera mode as needed would be a nice addition to the 3d viewer.

mp3guy, 2024-04-06:
This would be great. FWIW, the Pangolin mouse handler really is the gold standard in 3D: https://github.com/stevenlovegrove/Pangolin/blob/master/components/pango_display/src/handler.cpp#L238. It allows in-camera-plane translation of a 3D point under the cursor, rotation around a 3D point under the cursor, zoom towards/away from a 3D point under the cursor and rolling around the forward vector of the camera. Nothing comes close to this in terms of ease of use and power; MeshLab, Blender, cloudcompare etc.. all fall short.

Dvad, 2024-06-18:
Pangolin mouse handler is hard to beat.

Blender with those settings is also quite good:
[image]

Rotation and zooming around the 3d point under the cursor is a superpower that makes navigating into big scene a breeze.

It would be great to have something like this.