rerun-io/rerun #5835: Can't completely remove group in blueprint panel

nikolausWest, 2024-04-08:
Found on web version of the 0.15 release branch [here](https://www.rerun.io/viewer/pr/5816?url=https%3A%2F%2Fapp.rerun.io%2Fversion%2F0.15.0-rc.2%2Fexamples%2Fhuman_pose_tracking.rrd)

See video
https://github.com/rerun-io/rerun/assets/2624717/9830cc5d-4956-417b-9b40-5e947568e7c4

Also: hovering the entity still shows the old message that refers to groups, which is confusing.

abey79, 2024-04-08:
So we decided to _always_ show the origin, regardless of whether it shows up in data result or not. But clearly the `—`button shouldn't be there. Taking a look 👀

nikolausWest, 2024-04-08:
> So we decided to always show the origin, regardless of whether it shows up in data result or not. But clearly the —button shouldn't be there. Taking a look 👀

Alright, I didn't realize that. It's pretty confusing but I can see why you made that decision. Important that the tooltip is descriptive in that case as well then

closed 2024-04-08 by abey79