rerun-io/rerun #5862: Improve dependency management for our examples

abey79, 2024-04-08:
Managing dependencies for our growing list of example is increasingly painful, eg. https://github.com/rerun-io/rerun/issues/5859

One way about it would be to pin everything in the requirements.txt. This is some drawbacks:
- Updating pins is a painful, manual process, especially to handle incompatibilities between individual dependencies...
- ...yet we must do it on a regular basis, if only to support newer version of Python every year.
- Our approach of "merging" all of our examples' requirements.txt files makes the dependency compatibility problem even worse.
- Even with the perfect pins, this still doesn't give us a lock file, so things can still break via updates to transitive dependencies.

Ideally, we should have a system that allows for:
- Per example dependency definition.
- Per example lock file.
- Per example isolated environment.
- "Pythonic" (aka. compatible with  pip, PEP 517, etc.)
- Handle well at least the following two configuration:
  - dev environment where `rerun-sdk` should be installed from the workplace
  - user checking out example and running against a "regular" install of rerun (e.g. with `pipx`, etc.)

emilk, 2024-04-10:
I think we should looks for a solution that is `pixi` compatible:
* Any required tools can be installed and versioned from pixi

emilk, 2024-04-10:
We should also strongly consider speed. The "pip install" steps on CI often take 1-2 minutes, which is crazy. Hopefully `uv` would be a lot faster.

abey79, 2024-04-10:
So, for "normal" Python user:

```
pipx install pixi
pixi run extract-normally-expected-python-files-for some_example
cd examples/python/some_example
# then do whatever with the newly created requirement/pyproject files
```

This is a *significant* regression on the workflow, on the altar of:
- "using a single tool"
- caching (which there are other ways to do the same, e.g. by migrating from `pip` to `uv` in CI: https://github.com/astral-sh/uv?tab=readme-ov-file#dependency-caching, https://github.com/actions/setup-python/pull/818)

teh-cmc, 2024-04-10:
> So, for "normal" Python user:
> 
> ```
> pipx install pixi
> pixi run extract-normally-expected-python-files-for some_example
> cd examples/python/some_example
> # then do whatever with the newly created requirement/pyproject files
> ```

No, we would just commit those requirement files.

The path is either:
```sh
cd examples/python/urdf_thing
poetry main.py  # uses autogenerated requirements
```

or:
```sh
pip install pixi
pixi run urdf_thing
```

abey79, 2024-04-10:
> ```shell
> cd examples/python/urdf_thing
> poetry main.py  # uses autogenerated requirements
> ```

That is just not how it works 🤔  Anyway, very hard to discuss these thing without actually prototyping the thing and getting our priorities straight.