rerun-io/rerun #5883: Rerun TCP protocol should be more structured and only log warnings if the incoming connection is from a Rerun SDK

jleibs, 2024-04-09:
Currently, the Rerun TCP protocol immediately starts with a version number.

Unfortunately, our port, 9876 is fairly low-entropy and likely to show up in the wild from non-Rerun sources or as part of security probes. This means we are quite likely to see incoming connections from non-Rerun sources.

Currently we Warn on protocol version mismatch:
```
[2024-04-09T11:23:43Z WARN  re_sdk_comms::server] Closing connection to client at 130.237.72.38:58008: SDK client is using a newer protocol version (790) than the SDK server (0)
```

However, sometimes these incoming connections are outside of our control.  The more principled approach would be to ALWAYS start with a rerun-unique header and then follow that with the version number. This way if the expected header isn't there we can ignore the connection and not produce the warning.

gurry, 2024-05-07:
@emilk In case you haven't started, I have already created this fix.

Wumpf, 2024-05-16:
I messed up unfortunately and removed this fix last-minute from 0.16 which as of the time of writing is in the process of being released by our CI job.
Documented here https://github.com/rerun-io/rerun/pull/6332/files#r1603677569

PR #6253 (2024-05-14): Send TCP protocol header to ignore non-rerun clients

### What
Now we send a special header in the TCP connection before sending the protocol version. It helps the server recognize if the client is a rerun client or something else.

**Edit**
Have changed the approach to send the header _after_ the protocol version. See the discussion in comments.

* Fixes #5883 

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and the [Code of Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've include [... cut]
commit b4bd7e09 (2024-05-14): Send TCP protocol header to ignore non-rerun clients (#6253)

### What
Now we send a special header in the TCP connection before sending the
protocol version. It helps the server recognize if the client is a rerun
client or something else.

**Edit**
Have changed the approach to send the header _after_ the protocol
version. See the discussion in comments.

* Fixes #5883 

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've [... cut]
closed 2024-05-17 by Wumpf