rerun-io/rerun #5884: "Save blueprint" UI and a bit confusing in web / dark mode

ralfbiedert, 2024-04-09:
**Describe the annoyance**

Minor nit: I just tried 0.15 in my browser (FF 124, dark UI) and was playing with blueprints. At first I tried to open a recording, and the UI looks like so:

[image]

So my mind is being conditioned like _clearly the interactable UI elements are "dark buttons"_, and that works as expected. A little later I then tried to save a blueprint where the UI looks like this:

[image]

My intuitive reaction was to click "Ok", and I was a bit surprised when the dialog just disappeared. At first I assumed it already downloaded the file without me noticing, but it turns out I had to literally click into the gray text. 

The same issue is also present in the "Save recording ..."  dialog. 

**Suggestion**

I think the clearest and most efficient way (in terms of intent and click count) would be having 2 buttons, one that downloads and closes the modal, and the other one that just closes the modal (e.g., named "Download" or  "Save", and "Cancel"). That would also be consistent with how the (native Windows) desktop dialog is labeled:

[image]

**Desktop (please complete the following information):**
 - OS: Browser, Firefox 124.0.2 on Windows 11, Dark Mode

emilk, 2024-04-10:
Thanks for the issue!

This dialog is created by [rfd](https://crates.io/crates/rfd). We should consider replacing it with our own Rerun-themed dialog, or make a PR to `rfd` to improve theirs (e.g. to make it CSS stylable).

However, there is a second problem I'd like to fix with this dialog: it requires too many clicks. The "click here to download your file" dialog should be unnecessary. I _believe_ its only there because some browsers only allow you to download files as a response to a user action (a click), and `rfd` is made to work also if a background process spawns the file dialog. But, we should be able to fix that since we only spawn the dialog on clicks (or keyboard shortcuts), though this might require some small changes to `eframe` (to call update on each click, not just on `requestAnimationFrame`).

abey79, 2024-04-10:
Regarding CSS styling of rfd overlay, it seems that it uses fairly well defined ids:

[image]

However, it does emit corresponding inline CSS in a `<style>` tag. It would be nice to have an option for rfd to skip that, but I believe we could readily override this in our external CSS by using `!important`.